projf/projf-explore #14: Where does ‘data’ get its value from in sprite v1

jeremyherbert, 2021-01-09:
Hi,

In this line:

https://github.com/projf/projf-explore/blob/f6ed942204d9c33479b820e3aadc35adc9951fee/hardware-sprites/sprite_v1.sv#L70

the variable ‘data’ is referenced as containing the pixel colour. However, I can’t work out how that variable is initialised with a value. Could you please help me understand what is going on here? (I’ve read the file multiple times, I’m hoping I didn’t just overlook something obvious)

jeremyherbert, 2021-01-09:
Ah, I see now, it is attached via the wildcard in the ‘rom_async’ setup. I haven’t come across that syntax before.

closed 2021-01-09 by jeremyherbert

WillGreen, 2021-01-09:
Sorry, "data" isn't the best name. I recently switched to an explicit ROM for this data to handle a bug in Vivado and used generic names. I'll make the names clearer in the next update.

WillGreen, 2021-01-09:
I've updated the sprite examples with clearer signal names:

```verilog
    // sprite graphic ROM
    logic [$clog2(DEPTH)-1:0] spr_rom_addr;  // pixel position
    logic spr_rom_data;  // pixel colour
    rom_async #(
        .WIDTH(1),  // 1 bit per pixel
        .DEPTH(DEPTH),
        .INIT_F(SPR_FILE)
    ) spr_rom (
        .addr(spr_rom_addr),
        .data(spr_rom_data)
    );
```

The revised designs have been merged and published to the blog: https://projectf.io/posts/hardware-sprites/

jeremyherbert, 2021-01-09:
Thank you! And thank you for writing this, it is very helpful.