projf/projf-explore #16: Simulating clock_gen and getting only two pulses

jmlaniel, 2021-02-01:
Using Vivado 2020.2 on WIn 10, I downloaded the clock_gen and clock_gen_tb in order to simulate the clock generator for the XC7 FPGA. I am using a Basys 3 board.

I juste synthesize the file, then run the simulation. When I look at the output, I get only two pulse out of clk_480p and then it stops.

Is this normal? I'm not sure what I'm missing. Any help would be appreciated. I am learning FPGA, so there is a good chance that the problem has been caused by me.

[image]

WillGreen, 2021-02-01:
This is normal. It takes some time to generate a stable clock frequency.

If you scroll the simulation to the right, you should find the ~25 MHz clock starts running consistently.

By default, Vivado runs simulations for 1000ns. If you don’t see a locked clock by 1000ns, you can run it for additional time by clicking the simulation play button under “Help” near the top of the screen.

jmlaniel, 2021-02-01:
Thanks for your fast reply! 

It worked when I run the simulation a lot longer!

By the way, I really appreciate your website and the clarity of your explanation. If I may, I would like to suggest that you publish a page explaining what is going on in clock_gen. I don't understand it completely and the level of programming is quite high (especially for a beginner like me).

closed 2021-02-01 by jmlaniel

WillGreen, 2021-02-02:
Thank you. Covering clock synthesis in more detail is a good idea.