projf/projf-explore #34: Request to use your tutorial in a course

megpaciaroni, 2021-04-01:
Hi Will

Thank you very much for your very well written tutorial Hello Arty - Part 1. I don't usually do this, but I would like to go ahead and use your tutorial - as is - in my digital electronics class next week. I love you how you explain all the steps and the terminology so that it can be easily understood by almost anyone; I am sure my freshmen EE students will get a lot more out of following your tutorial than anything I can put together. But, I would like to have your permission first. Of course, I will let the students know who wrote this tutorial as I plan on just giving them the link to your page. Please let me know if you approve. Best and Thanks again! There's a lot out there and no one breaks it down like you did :-)

WillGreen, 2021-04-07:
Hello,

Thank you for your kind words. 

I don't mind you using _Hello Arty - Part 1_ as part of your course, provided it's for a single institution and you're not repackaging my content to sell online. It would be good to know where you're teaching.

I plan to make the third, and final, part of Hello Arty available later in the spring.

If you, or your students, have any suggestions or improvements please feel free to open issues on GitHub.

Best wishes,
Will

megpaciaroni, 2021-04-15:
Hi Will - definitely just for one class this semester. I learned digital electronics in the early 80s, so I am pretty weak with HDL! I am teaching at Benedictine College. I broke my leg this semester, so class preparation was severely hampered. Therefore, I was so grateful when I came across your tutorial and realized my freshmen could easily follow it. They did great with all 6 groups completing in 1 lab period. It was definitely a great springboard because they said they feel prepared to move on with their own designs. So again Thanks so much!!

megpaciaroni, 2021-04-15:
Oh and of course I will pass it on if they have any suggestions!

WillGreen, 2021-04-16:
Thanks for sharing and best wishes for your course.

closed 2021-04-16 by WillGreen