projf/projf-explore #98: Feedback on endianness

goekce, 2022-03-24:
On page https://projectf.io/posts/numbers-in-verilog/ you mention:

> Say you’ve got a bit-endian ...

You probably mean big

> ... byte from I2C and want to convert it to little-endian. Alas, you can’t mix big and little-endian vectors, so the following won’t work:

I would pay attention to endianness in this context. In my experience endianness is mostly known in data consisting of multiple bytes. I think you are talking about the endianness of bits, which seem to be called *bit endianness* ([example](https://stackoverflow.com/questions/6043483/why-bit-endianness-is-an-issue-in-bitfields).

---

Another issue I stumbled upon:

> wire [0:7] i2c_byte;  // 8-bit wire (big-endian)
> reg  [7:0] le_byte;   // 8-bit reg (little-endian)

> always_ff @(posedge clk) le_byte <= i2c_byte;  // Won't work :(

I tried:

```verilog
`define LED_COUNT 16

module led_sw_all_reversed(
  input [`LED_COUNT-1:0] sw,
  output [0:`LED_COUNT-1] led
);

assign led = sw;

endmodule
```

Which worked fine using Vivado on my hardware with 16 switches and LEDs. Not but least, I am new to Verilog.

PS: 😎 way of teaching FPGAs using graphics!

WillGreen, 2022-04-10:
Thank you for your suggestions. I'll review them in detail when I return to working on my _Maths and Algorithms with FPGAs_ series later this spring.

WillGreen, 2022-12-16:
Hello @goekce,

In the last couple of months, I've published a revised intro to numbers and a separate draft post covering vectors:

* https://projectf.io/posts/numbers-in-verilog/
* https://projectf.io/posts/verilog-vectors-arrays/

I hope these revised posts cover your suggestions, but I'd welcome any further feedback you have.

I appreciate your patience and good luck with your hardware designs.

goekce, 2022-12-16:
I don't know what you exactly changed, but I see that you removed endianness regarding bits. That looks ok to me!

closed 2022-12-16 by goekce