projf/projf-explore #127: Exporling ethernet on Arty A7

eliquinox, 2022-10-29:
Hi @WillGreen 

First of all thank you for your blog and github repos. They have so far been instrumental in my FPGA learning process. I want to get familiar with ethernet and create a project whereby I would blink leds every time the board clocks an ethernet packet with certain content.

Do you have any recommendations as to where to start? Would you be interested in publishing such a walkthrough on your blog and providing supplementary github materials? 

Ed

WillGreen, 2022-10-31:
Hello Ed, and thank you for your support. 

I don't have any plans to cover Ethernet in the immediate future.

However, I have three suggestions to get you going:

1. Most like a Project F approach is the fpga4fun.com Ethernet receiver series: https://www.fpga4fun.com/10BASE-T0.html, which doesn't use an external PHY IC or even require an RJ45 connection.

2. The Arty A7 has an Ethernet PHY and RJ45 jack, so an alternative approach would be to use the on-board PHY as described by https://digilent.com/reference/programmable-logic/arty-a7/reference-manual#ethernet_phy and the Texas Instruments datasheet: https://www.ti.com/product/DP83848J

3. Finally, www.fpga-cores.com has an Ethernet core in VHDL that's free for non-commercial or evaluation purposes and has instructions for the Arty A7: https://www.fpga-cores.com/tutorials/ethernet-on-arty-a7-board/ - I've not used this core, so can't comment or how easy-to-use or capable it is.

I hope this helps, and good luck with your hardware design,
Will

WillGreen, 2022-12-16:
Hi @eliquinox, I'll close this issue shortly unless you have any questions about my suggestions.

closed 2023-04-12 by WillGreen