projf/projf-explore #131: UART echo's back half the input

eliquinox, 2022-11-27:
I have set up the UART example with Arty A7 100 board and I observe the following:

When I send data to the USB descriptior as:

```
echo "AAAA" > /dev/ttyUSB3
``` 

In the minicom I can see 

```
AA
```
being output.

Is that expected? Shouldn't all the input be echo'd back or am I missing something?

eliquinox, 2022-11-28:
Hi @WillGreen any ideas about this?

WillGreen, 2022-12-01:
Thanks for your issue report. Can you tell me what serial settings you’re using?

NB. I’m away until next week. I’ll look into your issue when I return.

eliquinox, 2022-12-02:
Here are my `minicom` settings:
[image]
Let me know if you need other info.

WillGreen, 2022-12-09:
I do most of my Xilinx FPGA dev on Windows. I've used PuTTY for serial testing, which doesn't have this issue.

However, I've managed to reproduce your issue in Windows under WSL running Debian 11.

I configured the serial port (COM4):
```stty -F /dev/ttyS4 9600 cs8 -cstopb -parenb```

Then used cat and echo on the serial port and only saw every other character.

I don't yet know why this is happening I'm afraid. But will continue to investigate.

WillGreen, 2022-12-16:
Thanks for your reply. I'll leave the issue open until I get a chance to test this.
I can also try it on Linux with iCEBreaker board.

wil-low, 2023-05-28:
I've spend several hours to find reliable UART RX and TX implementation. The one that I use now is from Alinx AX301 demo board. I've added into my fork of your excellent repo: https://github.com/wil-low/projf-explore/tree/main/lib/uart-2

WillGreen, 2023-06-01:
> I've spend several hours to find reliable UART RX and TX implementation. The one that I use now is from Alinx AX301 demo board. I've added into my fork of your excellent repo: https://github.com/wil-low/projf-explore/tree/main/lib/uart-2

Cool. I’ve not looked at the UART in a while, but I plan to come back to it later this year as part of an FPGA design I’m working on.